waymo-research/waymax #1: Python 3.9-style type hints with `python_requires='>=3.8'`

schmrlng, 2023-10-20:
Congratulations on (and thank you for) the release! Just FYI, the [PEP 585](https://peps.python.org/pep-0585/)-style type hints used throughout this repo (e.g., [`tuple[int, ...]`](https://github.com/waymo-research/waymax/blob/main/waymax/config.py#L85) instead of the `from typing import Tuple` version) aren't compatible with Python 3.8. If you plan to support Python 3.8 I'd be happy to make a PR fixing all instances of such hinting (though it's probably more expediently handled on your end), or alternatively looking forward will you plan to only support >=3.9? For what it's worth, upstream JAX has already [given up on Python 3.8](https://github.com/google/jax/blob/main/CHANGELOG.md#jax-0414-july-27-2023) (and is even [considering dropping 3.9 soon](https://github.com/google/jax/pull/18072)), so the latter option wouldn't be too out-of-the-blue.

mcitir, 2023-10-20:
I have similar problem. 

> Congratulations on (and thank you for) the release! Just FYI, the [PEP 585](https://peps.python.org/pep-0585/)-style type hints used throughout this repo (e.g., [`tuple[int, [...]

justinjfu, 2023-10-23:
Thanks for bringing up this issue!

We've decided to only support Python 3.10+ for the time being, and I've updated the setup.py to reflect that. If there is significant interest in supporting 3.9 we can look into it, since it seems like the main issues are just (a) type hint syntax, and (b) JAX version support.

closed 2023-10-23 by justinjfu