waymo-research/waymax #3: The possibility of integrate multi-agent reinforcement learning libraries

fantastic8124, 2023-10-20:
Thank you for your excellent work！
From a practical application perspective, I wonder how to integrate the simulator with Marllib or Marl_ Benchmark and other multi-agent reinforcement learning libraries, like other exist simulator like metadrive?

justinjfu, 2023-10-20:
Hi @fantastic8124,

It should be possible to write a wrapper to other RL interfaces (we have several examples here: https://github.com/waymo-research/waymax/tree/main/waymax/env/wrappers). 

It looks like Marllib uses Ray's [MultiAgentEnv](https://docs.ray.io/en/latest/rllib/package_ref/env/multi_agent_env.html) interface, which has a sparse representation for agents - actions are passed in as a dict of {agent_key: action} and observations are returned as a dict of {agent_key: obs}. Waymax uses a dense representation by default (we expect and return arrays, with a valid mask), but it should be possible to convert between the two using an adapter.

We'll be busy supporting bugfixes and busy features for the time being with the initial release, but would be open to implementing this in the future or accepting a pull request with this feature.

samuela, 2025-05-30:
In the interest of cleaning up stale issues I will close this for now. But I agree with @justinjfu that this is interesting and PRs for it would be considered.

closed 2025-05-30 by samuela