waymo-research/waymax #5: All attempts to get a Google authentication bearer token failed

pascalbrunner, 2023-10-20:
Hi,

I logged in with my google account and did "gcloud init".
I gave all access via browser and got the console message:
You are now logged in as [MyName@googlemail.com] 

However I still get this message (in the meantime I also tried the other authentifications but unfortunately no success):

tensorflow/tsl/platform/cloud/google_auth_provider.cc:184] All attempts to get a Google authentication bearer token failed, returning an empty token. Retrieving token from files failed with "NOT_FOUND: Could not locate the credentials file.". Retrieving token from GCE failed with "FAILED_PRECONDITION: Error executing an HTTP request: libcurl code 6 meaning 'Couldn't resolve host name', error details: Could not resolve host: metadata.google.internal".

Any Suggestions?

pascalbrunner, 2023-10-20:
Update: I did it on another computer and it directly worked. I guess it could be because of my python or tensorflow version

justinjfu, 2023-10-20:
Possibly related to: https://github.com/tensorflow/datasets/issues/2761? The suggested fix was to upgrade the tensorflow verison.

In any case, I will close this issue since it looks like it's resolved. If the issue persists feel free to re-open.

closed 2023-10-20 by justinjfu