waymo-research/waymax #6: Error occurs in dataloading

BlueCat-de, 2023-10-20:
The listed error occurs while I'm running the following codes in python3.9:
```python
from waymax import config
from waymax import dataloader

scenarios = dataloader.simulator_state_generator(config.WOD_1_1_0_TRAINING)
scenario = next(scenarios)
```

```
Traceback (most recent call last):
  File "/mnt/203663f6-7995-4295-8c53-9e2e9621b487/cqm/Waymo_learn/waymo.py", line 2, in <module>
    from waymax import dataloader
  File "/home/qh802/anaconda3/envs/Waymo/lib/python3.9/site-packages/waymax/dataloader/__init__.py", line 16, in <module>
    from waymax.dataloader.dataloader_utils import get_data_generator
  File "/home/qh802/anaconda3/envs/Waymo/lib/python3.9/site-packages/waymax/dataloader/dataloader_utils.py", line 181, in <module>
    Callable[[bytes], dict[str, tf.Tensor | dict[str, tf.Tensor]]]
TypeError: unsupported operand type(s) for |: 'type' and 'types.GenericAlias'
```
Does the error occurs because of the python version I'm using?

MaisJamal, 2023-10-20:
I have a similar issue
```
from waymax import config
```

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/mais/waymax/waymax/config.py", line 39, in <module>
    class DatasetConfig:
  File "/home/mais/waymax/waymax/config.py", line 85, in DatasetConfig
    batch_dims: tuple[int, ...] = ()
TypeError: 'type' object is not subscriptable
```

justinjfu, 2023-10-20:
Thanks for bringing up this issue. The dict union syntax mentioned by @BlueCat-de should be available starting from Python 3.9 (https://peps.python.org/pep-0584/). The generic type hints mentioned by @MaisJamal should also be available in 3.9 (https://peps.python.org/pep-0585/)

I will perform some more testing with different python versions, but 3.11 should work for the time being. In the worse case, we will bump the required python version higher.

justinjfu, 2023-10-23:
Update: I've updated the setup.py to require Python 3.10 and above, and confirmed the code works with those versions.

Also see: https://github.com/waymo-research/waymax/issues/1

closed 2023-10-23 by justinjfu