waymo-research/waymax #10: Hello, I have encountered the following error message. How can I resolve it

like2000522, 2023-10-22:
2023-10-22 20:35:38.005846: W tensorflow/tsl/platform/cloud/google_auth_provider.cc:184] All attempts to get a Google authentication bearer token failed, returning an empty token. Retrieving token from files failed with "NOT_FOUND: Could not locate the credentials file.". Retrieving token from GCE failed with "FAILED_PRECONDITION: Error executing an HTTP request: libcurl code 6 meaning 'Couldn't resolve host name', error details: Could not resolve host: metadata.google.internal".

Jin0932, 2023-10-24:
Hi, I meet the same question in macos10.15, how do i solve?

like2000522, 2023-10-24:
> I had similar Google Auth/Permission Denied error on Windows. Exporting GOOGLE_APPLICATION_CREDENTIALS to the notebook and restarting the kernel fixed it.
> 
> `os.environ['GOOGLE_APPLICATION_CREDENTIALS'] =r'C:\Users\<USER-NAME>\AppData\Roaming\gcloud\application_default_credentials.json'`

Thank you for your reply, it is very important to me. I am currently trying to run through this emulator and have encountered new issues. May I ask if you have also encountered this issue：
tensorflow.python.framework.errors_impl.AbortedError: {{function_node __wrapped__IteratorGetNext_output_types_31_device_/job:localhost/replica:0/task:0/device:CPU:0}} All 10 retry attempts failed. The last failure: Error executing an HTTP request: libcurl code 7 meaning 'Couldn't connect to server', error details: Failed to connect to storage.googleapis.com port 443 after 2005 ms: Couldn't connect to server
	 when reading gs://waymo_open_dataset_motion_v_1_2_0/uncompressed/tf_example/training/training_tfexample.tfrecord-00000-of-01000 [Op:IteratorGetNext] name:

justinjfu, 2023-10-24:
Can you verify that you have permissions set up properly to access Google cloud storage? e.g. try accessing the dataset via gsutil:

```
$ gsutil ls gs://waymo_open_dataset_motion_v_1_2_0/uncompressed
gs://waymo_open_dataset_motion_v_1_2_0/uncompressed/lidar/
gs://waymo_open_dataset_motion_v_1_2_0/uncompressed/scenario/
gs://waymo_open_dataset_motion_v_1_2_0/uncompressed/tf_example/
```

Jin0932, 2023-10-25:
> gsutil ls gs://waymo_open_dataset_motion_v_1_2_0/uncompressed

Thanks, and yes, the result as follow:
 `
(waymo) MacBook-Pro:waymodemo jqw$ gsutil ls gs://waymo_open_dataset_motion_v_1_2_0/uncompressed
gs://waymo_open_dataset_motion_v_1_2_0/uncompressed/lidar/
gs://waymo_open_dataset_motion_v_1_2_0/uncompressed/scenario/
gs://waymo_open_dataset_motion_v_1_2_0/uncompressed/tf_example/
`

justinjfu, 2023-10-25:
@Jin0932 Hmmm that's weird. Have you also checked that you've created the application default credentials? e.g. On Linux, there should be a file called `~/.config/gcloud/application_default_credentials.json`

If that file doesn't exist, you need to run `gcloud auth application-default login`

closed 2025-05-30 by samuela