waymo-research/waymax #9: Loading Scenarios with Multi-GPU Support

Jynxzzz, 2023-10-22:
Hello,

I am exploring the use of Waymax for my autonomous driving research and am interested in utilizing multiple GPUs to load and simulate different scenarios. Could you provide guidance or examples on how this can be achieved? Is there built-in support for multi-GPU scenario loading, or are there recommended practices to follow when implementing this functionality?

Thank you for your assistance!

justinjfu, 2023-10-23:
Multi-GPU simulation can be done as follows: 
a) Pass in `distributed=True` into the `DatasetConfig` for the dataloader. This will add an additional batch dimension equal to the number of devices on your machine.
c) For any simulations you run, wrap your simulation in `jax.pmap`, which will distribute the computation across all devices.

This will only do multi-GPU simulation. I'm less familiar with multi-GPU data loading using tf.Datasets - you could maybe take a look at distributed TF data services (https://www.tensorflow.org/api_docs/python/tf/data/experimental/service/distribute) via the `tf_data_service_address` option on `DatasetConfig`.

Jynxzzz, 2023-10-24:
I am excited to inform you that your advice was incredibly helpful and I have successfully implemented the multi-GPU simulation and data loading following your suggestions. Everything is working smoothly now. I can't thank you enough for your guidance and for taking the time to assist me. Your expertise made a significant difference in resolving the challenges I was facing. Thank you once again!

closed 2023-10-24 by Jynxzzz