waymo-research/waymax #13: How can I get the HD map information from waymax?

InterKnight, 2023-10-23:
Hi, Thank you for great work ! ! !
I have a question on using waymax to test planning algorithm: 
I notic that “Waymax is designed to support research for closed-loop simulation for planning”, and I'm wondering if the planning algorithm need HD map, how can I get the map information from waymax and convert data format to meet algorithm requirements？

InterKnight, 2023-10-23:
Is waymax can be used for testing rule-based plannging algorithm? or only suit for IL & RL planning algorithm ?

justinjfu, 2023-10-23:
Hi @InterKnight,

> I'm wondering if the planning algorithm need HD map, how can I get the map information from waymax and convert data format to meet algorithm requirements？

At this point in time, we support only the map information contained in WOMD - see the roadgraph_samples field in https://waymo.com/open/data/motion/tfexample. This means you have access to information on road boundaries, lane boundaries, lane orientation, which lanes traffic lights are controlling, etc.

If you are using the dataloader, you can find the roadgraph information in SimulatorState.roadgraph_points. Let me know if you have more specific questions about a particular feature that you need.

> Is waymax can be used for testing rule-based plannging algorithm? or only suit for IL & RL planning algorithm ?

Waymax doesn't make any assumptions about the planning algorithm. The simulator only expects an action (X+Y+Yaw, Steering+Acceleration, etc.) that can be produced by any algorithm.

InterKnight, 2023-10-24:
Thank you！ @justinjfu

closed 2023-10-24 by InterKnight