waymo-research/waymax #14: Is this running speed normal?

jkewang, 2023-10-23:
First of all thank you for your great work!
I read in the paper, "As a point of reference, evaluating the full WOD evaluation dataset (44K scenarios) with 8-V100 machine takes less than 2min."
So I write a simple code to evaluate the running speed:
```
config = dataclasses.replace(_config.WOD_1_1_0_VALIDATION, max_num_objects=32)
data_iter = dataloader.simulator_state_generator(config=config)

scenario = next(data_iter)
start_time = time.time()

while scenario:
    state = scenario
    for _ in range(scenario.remaining_timesteps):
        state = datatypes.update_state_by_log(state, num_steps=1)
    print("remaining_timesteps: ", scenario.remaining_timesteps)
    scenario = next(data_iter)
    end_time = time.time()
    print("time cost: ", end_time - start_time)
    start_time = end_time
```

I downloaded the "uncompressed_tf_example_validation_validation_tfexample.tfrecord-00000-of-00150" before the test in case of the download time delay. The time consuming shows like:
```
2023-10-23 15:44:39.018309: I tensorflow/core/platform/cpu_feature_guard.cc:193] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations:  AVX2 AVX_VNNI FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-10-23 15:44:39.124359: I tensorflow/core/util/port.cc:104] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2023-10-23 15:44:40.742728: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2023-10-23 15:44:40.768102: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2023-10-23 15:44:40.768229: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2023-10-23 15:44:40.769646: I tensorflow/core/platform/cpu_feature_guard.cc:193] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations:  AVX2 AVX_VNNI FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-10-23 15:44:40.770238: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2023-10-23 15:44:40.770354: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2023-10-23 15:44:40.770407: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2023-10-23 15:44:41.477894: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2023-10-23 15:44:41.478016: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2023-10-23 15:44:41.478077: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2023-10-23 15:44:41.478141: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1613] Created device /job:localhost/replica:0/task:0/device:GPU:0 with 9233 MB memory:  -> device: 0, name: NVIDIA GeForce RTX 3060, pci bus id: 0000:01:00.0, compute capability: 8.6
2023-10-23 15:44:41.495533: I external/xla/xla/pjrt/tfrt_cpu_pjrt_client.cc:462] TfrtCpuClient created.
2023-10-23 15:44:41.496053: I external/xla/xla/stream_executor/cuda/cuda_gpu_executor.cc:894] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero. See more at https://github.com/torvalds/linux/blob/v6.0/Documentation/ABI/testing/sysfs-bus-pci#L344-L355
2023-10-23 15:44:41.496203: I external/xla/xla/service/service.cc:168] XLA service 0x55d8d29d1e50 initialized for platform CUDA (this does not guarantee that XLA will be used). Devices:
2023-10-23 15:44:41.496220: I external/xla/xla/service/service.cc:176]   StreamExecutor device (0): NVIDIA GeForce RTX 3060, Compute Capability 8.6
2023-10-23 15:44:41.496421: I external/xla/xla/pjrt/gpu/se_gpu_pjrt_client.cc:633] Using BFC allocator.
2023-10-23 15:44:41.496443: I external/xla/xla/pjrt/gpu/gpu_helpers.cc:105] XLA backend allocating 9454682112 bytes on device 0 for BFCAllocator.
WARNING:tensorflow:From /home/conda/envs/waymo/lib/python3.10/site-packages/tensorflow/python/autograph/pyct/static_analysis/liveness.py:83: Analyzer.lamba_check (from tensorflow.python.autograph.pyct.static_analysis.liveness) is deprecated and will be removed after 2023-09-23.
Instructions for updating:
Lambda fuctions will be no more assumed to be used in the statement where they are used, or at least in the same block. https://github.com/tensorflow/tensorflow/issues/56089
2023-10-23 15:44:43.223023: I external/xla/xla/stream_executor/cuda/cuda_dnn.cc:440] Loaded cuDNN version 8904
2023-10-23 15:44:43.667560: I external/xla/xla/stream_executor/cuda/cuda_asm_compiler.cc:52] Using nvlink for parallel linking
remaining_timesteps:  90
time cost:  1.4673659801483154
remaining_timesteps:  90
time cost:  1.016791820526123
remaining_timesteps:  90
time cost:  1.1033782958984375
remaining_timesteps:  90
time cost:  1.0560741424560547
remaining_timesteps:  90
time cost:  1.0841765403747559
remaining_timesteps:  90
time cost:  1.241203784942627
remaining_timesteps:  90
time cost:  1.0544793605804443
remaining_timesteps:  90
time cost:  1.050783395767212
remaining_timesteps:  90
time cost:  1.061511516571045
remaining_timesteps:  90
time cost:  1.043912410736084
```

My GPU is NVIDIA GeForce RTX 3060, Compute Capability 8.6, cuda version is  cuda_11.6. After testing, Each 90-step case simulation takes approximately 1 second and the main time consumption is in this line of code:
```
state = datatypes.update_state_by_log(state, num_steps=1)
```

I'd like to know if this is running normally, or is there something wrong with my jax settings that's slowing it down?

justinjfu, 2023-10-23:
We usually run our code with jit. Could you try a snippet like this on your machine? I'm guessing it will be much faster.

```
config = dataclasses.replace(_config.WOD_1_1_0_VALIDATION, max_num_objects=32)
data_iter = dataloader.simulator_state_generator(config=config)

scenario = next(data_iter)
start_time = time.time()

@jax.jit
def benchmark(state):
  for _ in range(state.remaining_timesteps):
        state = datatypes.update_state_by_log(state, num_steps=1)
  return state

while scenario:
    state = scenario
    state = benchmark(state)
    print("remaining_timesteps: ", scenario.remaining_timesteps)
    scenario = next(data_iter)
    end_time = time.time()
    print("time cost: ", end_time - start_time)
    start_time = end_time
```

jkewang, 2023-10-24:
Yes, you are right. After I tried using @jax.jit, the loop that previously took 1 second only took about 0.007 seconds, which was 150 times faster.

Thank you for your great advice!

closed 2023-10-24 by jkewang

the-big-bad-wolf, 2025-02-04:
> We usually run our code with jit. Could you try a snippet like this on your machine? I'm guessing it will be much faster.
>
> ```
> config = dataclasses.replace(_config.WOD_1_1_0_VALIDATION, [...]

Traceback (most recent call last):
  File "-:\---\---\---\---\main.py", line 38, in <module>
    state = benchmark(state)
            ^^^^^^^^^^^^^^^^
  File "-:\---\---\---\---\main.py", line 30, in benchmark
    for _ in range(state.remaining_timesteps):

jax.errors.TracerIntegerConversionError: The __index__() method was called on traced array with shape int32[]
The error occurred while tracing the function benchmark at c:\Users\---\Documents\WaymaxAutoPlanner\main.py:28 for jit. This concrete value was not available in Python because it depends on the value of the argument state.timestep.
See https://jax.readthedocs.io/en/latest/errors.html#jax.errors.TracerIntegerConversionError

Tried inputting the timesteps as a static arg, but that also crashed with the error:
ValueError: Non-hashable static arguments are not supported. An error occurred while trying to hash an object of type <class 'jaxlib.xla_extension.ArrayImpl'>, 90. The error was:
TypeError: unhashable type: 'jaxlib.xla_extension.ArrayImpl'

The type hints tell me that state.remaining_timesteps is an int, so not sure why it is an array