waymo-research/waymax #16: Is this framework capable with multi-gpu simulation or managed by Ray?

Sephirex-X, 2023-10-24:
Hi, thanks for this amazing effort.
I'm wondering if your guys can give some more examples about how to do RL under your framework by either multi-gpu simulation or Ray, if it's possible? 
thx a lot

Sephirex-X, 2023-10-24:
BTW, is it possible to provide some examples about how to load WOMD locally, since all the configs provided are based on google cloud, many thx!

justinjfu, 2023-10-24:
Hi @Sephirex-X,

Thanks for the question - we're gathering feedback like this and will add additional documentation based on what users are having questions about.

For now - you can do multi-GPU simulation by (a) passing `distributed=True` to the dataset function which will add an additional batch dimension to the data, and (b) using the `jax.pmap` decorator to distribute across multiple GPUs.

Basically, your code will have this structure:
```
data_config = dataclasses.replace(_config.WOD_1_1_0_TRAINING, distributed=True)
dataset = dataloader.simulator_state_generator(config=config)

@jax.pmap
def simulate(scenario: datatypes.SimulatorState):
   ... your code here ...

simulate(next(dataset))
```

Related issue: https://github.com/waymo-research/waymax/issues/9

justinjfu, 2023-10-24:
For your other questions:

We haven't experimented with Ray yet. In my opinion a good path to integrate would be to write a wrapper to convert Waymax to Ray's environment API, similar to what we already have for DMEnv (https://github.com/waymo-research/waymax/tree/main/waymax/env/wrappers), but this is not on our timeline at the moment.

> BTW, is it possible to provide some examples about how to load WOMD locally, since all the configs provided are based on google cloud, many thx!

Tou can pass in a local path to the dataset config to use a local dataset (see https://github.com/waymo-research/waymax/blob/main/waymax/config.py#L39)

```
data_config = DatasetConfig(
    path='local_path/abc/xyz.tfrecord@1000',
    max_num_rg_points=20000,
    data_format=DataFormat.TFRECORD,
)
```

Sephirex-X, 2023-10-25:
Thank you every much!

closed 2023-10-25 by Sephirex-X

Sephirex-X, 2023-10-26:
> Hi @Sephirex-X,
>
> Thanks for the question - we're gathering feedback like this and will add additional documentation based on what users are having questions about.
>
> For now - you can do multi-GPU [...]

hi when i run this(single GPU), i got huge mem consume but low efficiency, how can i improve the computation efficiency?
[image]