waymo-research/waymax #18: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.

XiaomuWang, 2023-10-25:
I have already stored Waymo 1.2 data locally, and the data is located at the following path:
`-/workspace/root/waymo_open_dataset_motion_v_1_2_0/uncompressed/scenario/testing/
--testing.tfrecord-00000-of-00150
--testing.tfrecord-00001-of-00150
--...
--testing.tfrecord-00150-of-00150
`
When I run the following code to load local data:
`import numpy as np
import mediapy
from tqdm import tqdm
import dataclasses

from waymax import config as _config
from waymax import dataloader
from waymax import datatypes
from waymax import visualization

config = _config.DatasetConfig(path = '/workspace/root/waymo_open_dataset_motion_v_1_2_0/uncompressed/scenario/testing/testing.tfrecord@150',max_num_rg_points=30000,data_format=_config.DataFormat.TFRECORD)
# config = _config.DatasetConfig(path = '/workspace/datasets/waymo/training_20s/uncompressed_scenario_training_20s_training_20s.tfrecord@1000',max_num_rg_points=30000,data_format=_config.DataFormat.TFRECORD)

data_iter = dataloader.simulator_state_generator(config=config)
scenario = next(data_iter)`
I encounter the following error:
`(waymax) root@d070113b7ff4:/workspace/wangs/agent/waymax/docs# python test.py 
An NVIDIA GPU may be present on this machine, but a CUDA-enabled jaxlib is not installed. Falling back to cpu.
2023-10-25 15:27:31.911080: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.911261: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.911451: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.911726: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
Traceback (most recent call last):
  File "/workspace/wangs/agent/waymax/docs/test.py", line 15, in <module>
2023-10-25 15:27:31.912018: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
    scenario = next(data_iter)
               ^^^^^^^^^^^^^^^
  File "/opt/conda/envs/waymax/lib/python3.11/site-packages/waymax/dataloader/dataloader_utils.py", line 229, in get_data_generator
2023-10-25 15:27:31.912232: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
    for example in dataset.as_numpy_iterator():
  File "/opt/conda/envs/waymax/lib/python3.11/site-packages/tensorflow/python/data/ops/dataset_ops.py", line 4662, in __next__
2023-10-25 15:27:31.912606: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.912689: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.912813: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
    return nest.map_structure(to_numpy, next(self._iterator))
                                        ^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/envs/waymax/lib/python3.11/site-packages/tensorflow/python/data/ops/iterator_ops.py", line 797, in __next__
    return self._next_internal()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/envs/waymax/lib/python3.11/site-packages/tensorflow/python/data/ops/iterator_ops.py", line 780, in _next_internal
    ret = gen_dataset_ops.iterator_get_next(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/envs/waymax/lib/python3.11/site-packages/tensorflow/python/ops/gen_dataset_ops.py", line 3016, in iterator_get_next
    _ops.raise_from_not_ok_status(e, name)
  File "/opt/conda/envs/waymax/lib/python3.11/site-packages/tensorflow/python/framework/ops.py", line 7262, in raise_from_not_ok_status
2023-10-25 15:27:31.914974: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
    raise core._status_to_exception(e) from None  # pylint: disable=protected-access
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
tensorflow.python.framework.errors_impl.InvalidArgumentError: {{function_node __wrapped__IteratorGetNext_output_types_31_device_/job:localhost/replica:0/task:0/device:CPU:0}} **Feature: roadgraph_samples/dir (data type: float) is required but could not be found.**
         [[{{node ParseExample/ParseExampleV2}}]] [Op:IteratorGetNext]
2023-10-25 15:27:31.919280: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.919832: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.922848: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.923284: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.924494: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.925009: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.932046: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.932129: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.935198: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.943724: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.944583: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.946058: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
[... 31 lines cut ...]
2023-10-25 15:27:31.960596: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
[... cut ...]
2023-10-25 15:27:31.980400: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.
2023-10-25 15:27:31.982069: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: roadgraph_samples/dir (data type: float) is required but could not be found.`
I suspect it's a TensorFlow version issue. I've tried both 2.12.0 and 2.14.0, but it didn't work.

justinjfu, 2023-10-25:
@XiaomuWang To narrow down where the issue is - does using the remote google cloud path work? (use `gs://waymo_open_dataset_motion_v_1_2_0/uncompressed/tf_example/testing/testing_tfexample.tfrecord@150`) That way we can verify if it's a problem with the dataset versus a problem with the installation setup.

Tensorflow 2.14.0 should work. I have that installed on my local machine and I can properly load the dataset from google cloud.

XiaomuWang, 2023-10-26:
Thank you for your quick response. I have identified the issue. The problem arose because I was using data from the "scenario," which differs from the "tf_example" data. When I loaded the "tf_example" data locally, there were no issues.

closed 2023-10-26 by XiaomuWang