waymo-research/waymax #19: How to Select Specific Scenarios in Waymax Based on WAYMO OPEN MOTION DATASET Methodology?

Jynxzzz, 2023-10-26:
I am exploring the utilization of Waymax for simulating scenarios based on the Waymo's data. I recently came across a paper titled "Large Scale Interactive Motion Forecasting for Autonomous Driving: The WAYMO OPEN MOTION DATASET", where the authors outlined a method to mine specific pairwise interaction scenarios from a large dataset. They mentioned:
"we specifically mined for the following pairwise interaction scenarios: merges, lane changes, unprotected turns, intersection left turns, intersection right turns, pedestrian-vehicle interactions, cyclist-vehicle interactions, interactions with close proximity, and interactions with high accelerations.”

I am intrigued to understand how I could adopt a similar approach within Waymax to select specific scenarios like intersections or merging scenarios. Is there a recommended method or tool within Waymax that facilitates the selection or annotation of these specific interaction scenarios?

I am particularly interested in automating the process of scenario selection based on certain criteria as outlined in the paper, and I'm looking for any guidance on how to approach this within the Waymax environment.

Here's the excerpt from the paper for reference:
"we specifically mined for the following pairwise interaction scenarios: merges, lane changes, unprotected turns, intersection left turns, intersection right turns, pedestrian-vehicle interactions, cyclist-vehicle interactions, interactions with close proximity, and interactions with high accelerations.”

Thank you!

justinjfu, 2023-10-30:
My understanding of this question is that you would like to filter the dataset elements by some criteria, and then write them back to disk?

You could try the following strategy:
1. Use the dataloader to read TF examples as raw dictionaries (https://github.com/waymo-research/waymax/blob/main/waymax/dataloader/dataloader_utils.py#L63)
2. Write a filter function to match the criteria of your choice. Inside the filter function you could use the Waymax datastructures if you find those easier to work with, but it's optional. To convert dictionaries to Waymax datastructures use the provided factory functions in: https://github.com/waymo-research/waymax/blob/main/waymax/dataloader/womd_factories.py#L135
3. Convert the results back to TF record format and write to disk by following this example: https://www.tensorflow.org/tutorials/load_data/tfrecord.

johannes-fischer, 2024-06-14:
Did anyone have success in filtering scenarios based on their type and has some advice or can share code? I am for instance interesting in merging/lane changing scenarios, but also the other interactive types mentioned in the first post.