waymo-research/waymax #21: How can run simluation on ego-car

Sephirex-X, 2023-10-26:
Hi, after checking your doc I found `class waymax.env.planning_agent_environment.PlanningAgentEnvironment`. But the tutorials only provide the multi-agents version, could you provide some examples on how to implement this method? 
Thank you sooooo much.

Jin0932, 2023-11-30:
@Sephirex-X  hi， Can you now use the planning_agent_environment environment? If so, please reply and let’s talk about it together, thank you

Ananasburn, 2024-12-21:
> @Sephirex-X hi， Can you now use the planning_agent_environment environment? If so, please reply and let’s talk about it together, thank you

hallo! Do you know already how to run simulation for controlling the ego car?    I would be so appreciated for your reply :D

Ananasburn, 2025-01-05:
> > > @Sephirex-X hi， Can you now use the planning_agent_environment environment? If so, please reply and let’s talk about it together, thank you
> > 
> > 
> > hallo! Do you know already how to run simulation for controlling the ego car? I would be so appreciated for your reply :D
> 
> Hi, I manage to do so by using mulit-agent env, control the ego one using your policy and others using the log-traj.

Hallo! Thanks a lot for your information!  I'll try to do so.

Sephirex-X, 2025-01-06:
> Hallo! Thanks a lot for your information! I'll try to do so.

You can also check my latest repo, I have managed to provide a complete starter kit.

Ananasburn, 2025-01-07:
> You can also check my latest repo, I have managed to provide a complete starter kit.

Awesome! I have already checked this repo but i'm confused about where did you setup the ego actor and the others. Could you explain it a bit?

closed 2025-01-08 by Sephirex-X