waymo-research/waymax #24: Minimum system requirement

mshah72, 2023-10-27:
What is the minimum system requirement for this simulator?

Is it possible to try this using Google Colab?

justinjfu, 2023-10-31:
Hi,

We haven't done extensive testing on different machines for minimum requirements. A modern GPU or TPU will speed up simulation significantly over any CPU (e.g. NVidia 3080+), and more GPU/TPU memory is helpful for larger batch sizes.

Google colab should work, just run `!pip install git+https://github.com/waymo-research/waymax.git@main#egg=waymo-waymax` at the top of the colab.` If you are a researcher this is excellent for being able to take advantage of the TPUs/GPUs that come with colab.

samuela, 2025-05-30:
In the interest of cleaning up stale issues I will close this for now.

closed 2025-05-30 by samuela