waymo-research/waymax #25: INVALID_ARGUMENT: Feature: path_samples/arc_length (data type: float) is required but could not be found.

Sephirex-X, 2023-10-27:
Hi, when I set my conifg like:
```Shell
conf = DatasetConfig(
    path='My_path_to_dataset',
    max_num_rg_points=20000,
    data_format=DataFormat.TFRECORD,
    max_num_objects=max_num_objects,
    num_shards = 48,
    repeat= 1,
    include_sdc_paths=True,
    num_paths= 1,
    num_points_per_path= 1,
)
```

I got this error:
```Shell
2023-10-27 16:50:13.885355: W tensorflow/core/framework/op_kernel.cc:1828] OP_REQUIRES failed at example_parsing_ops.cc:98 : INVALID_ARGUMENT: Feature: path_samples/arc_length (data type: float) is required but could not be found.
```
I located the occurance on `waymax/waymax/dataloader/dataloader_utils.py`,line229:
```Shell
for example in dataset.as_numpy_iterator():
      yield postprocess_fn(example)
```
It would be great if you can help me out there! Thx

justinjfu, 2023-10-27:
Hi @Sephirex-X,

This feature is meant to support routes for the SDC, but it's not public in the Waymo Open Dataset yet. We should have marked this flag with a "NotImplementedError".

Sephirex-X, 2023-10-28:
> Hi @Sephirex-X,
> 
> This feature is meant to support routes for the SDC, but it's not public in the Waymo Open Dataset yet. We should have marked this flag with a "NotImplementedError".
@justinjfu 
Yeah I see, thank u 4 ur instant response, it's a very great work, however there r few things trouble me a lot. 
1. How can I use the SDC reward provided by ur framework in this case? 
2. Is there any ways that I can use the expert log on SDC to do my own reward calculation? 
3. I'm really confused about the relationship between missed routes for SDC (used to calculate reward for SDC I suppose) and the logged expert trajectory for SDC, is it possible u can provided some detail description? 
4. And finally, based on the code and data provided right now, is it possible to run some RL algorithms on SDC.
Many thxs 4 ur time!

hz3014, 2024-08-15:
Hi @Sephirex-X , Did you find the answer to the questions?
I am also confused with the difference between simulator_state.sdc_paths and simulator_state.log_trajectory.

closed 2025-01-06 by Sephirex-X