waymo-research/waymax #29: Unable to access dataset

binshi, 2023-10-31:
Despite registering multiple times, the dataset cannot be accessed in gcloud. 
`DatasetConfig(path='gs://waymo_open_dataset_motion_v_1_0_0/uncompressed/tf_example/validation/validation_tfexample.tfrecord@150', data_format=<DataFormat.TFRECORD: 'TFRECORD'>, repeat=None, batch_dims=(), shuffle_seed=None, shuffle_buffer_size=1000, num_shards=4, deterministic=True, include_sdc_paths=False, aggregate_timesteps=True, max_num_rg_points=20000, max_num_objects=32, num_paths=None, num_points_per_path=None, drop_remainder=True, tf_data_service_address=None, distributed=False, batch_by_scenario=True)
---------------------------------------------------------------------------
PermissionDeniedError                     Traceback (most recent call last)
[<ipython-input-4-3585dad5ed34>](https://localhost:8080/#) in <cell line: 7>()
      5 print(config)
      6 data_iter = dataloader.simulator_state_generator(config=config)
----> 7 scenario = next(data_iter)

5 frames
[/usr/local/lib/python3.10/dist-packages/tensorflow/python/framework/ops.py](https://localhost:8080/#) in raise_from_not_ok_status(e, name)
   5886 def raise_from_not_ok_status(e, name) -> NoReturn:
   5887   e.message += (" name: " + str(name if name is not None else ""))
-> 5888   raise core._status_to_exception(e) from None  # pylint: disable=protected-access
   5889 
   5890 

PermissionDeniedError: {{function_node __wrapped__IteratorGetNext_output_types_31_device_/job:localhost/replica:0/task:0/device:CPU:0}} Error executing an HTTP request: HTTP response code 403 with body '<?xml version='1.0' encoding='UTF-8'?><Error><Code>AccessDenied</Code><Message>Access denied.</Message><Details>Anonymous caller does not have storage.objects.get access to the Google Cloud Storage object. Permission 'storage.objects.get' denied on resource (or it may not exist).</Details></Error>'
	 when reading gs://waymo_open_dataset_motion_v_1_0_0/uncompressed/tf_example/validation/validation_tfexample.tfrecord-00000-of-00150 [Op:IteratorGetNext] name:`

` Additional permissions required to list objects in this bucket. Ask a bucket owner to grant you 'storage.objects.list' permission. Learn more `

cjy875428, 2023-10-31:
we met the same problem,if you have a solution,please e-mail to junyuchen@bupt.edu.cn

binshi, 2023-11-03:
> Hi All,
> 
> Could you try some of the steps in [#30 (comment)](https://github.com/waymo-research/waymax/issues/30#issuecomment-1787765649) and see if they resolve the issue? And it would be appreciated if you report back if they work so we can compile an FAQ for these access issues.

Tried the steps, but still unable to access the datasets.

samuela, 2025-05-30:
In the interest of cleaning up stale issues I will close this for now. Feel free to open a new issue if you are still experiencing this!

closed 2025-05-30 by samuela