waymo-research/waymax #30: ERROR :All attempts to get a Google authentication bearer token failed, returning an empty token

DA21S321D, 2023-10-31:
I code as your tutorial says in https://waymo-research.github.io/waymax/docs/notebooks/data_demo.html . I even can't run the first two block Here is my simplest code that apprently enough to recognize:

import dataclasses
from waymax import config as _config
from waymax import dataloader

config = dataclasses.replace(_config.WOD_1_1_0_TRAINING, max_num_objects=32)
data_iter = dataloader.simulator_state_generator(config=config)
scenario = next(data_iter)

Here is error: 
F:\python_envirments\waymax\Scripts\python.exe C:\Users\13420\Desktop\waymax\22.py 
2023-10-31 18:46:30.457267: W tensorflow/tsl/platform/cloud/google_auth_provider.cc:184] All attempts to get a Google authentication bearer token failed, returning an empty token. Retrieving token from files failed with "NOT_FOUND: Could not locate the credentials file.". Retrieving token from GCE failed with "FAILED_PRECONDITION: Error executing an HTTP request: libcurl code 6 meaning 'Couldn't resolve host name', error details: Could not resolve host: metadata.google.internal".
Traceback (most recent call last):
  File "C:\Users\13420\Desktop\waymax\22.py", line 13, in <module>
    scenario = next(data_iter)
  File "C:\Users\13420\Desktop\waymax\waymax\dataloader\dataloader_utils.py", line 229, in get_data_generator
    for example in dataset.as_numpy_iterator():
  File "F:\python_envirments\waymax\lib\site-packages\tensorflow\python\data\ops\dataset_ops.py", line 4691, in __next__
    return nest.map_structure(to_numpy, next(self._iterator))
  File "F:\python_envirments\waymax\lib\site-packages\tensorflow\python\data\ops\iterator_ops.py", line 809, in __next__
    return self._next_internal()
  File "F:\python_envirments\waymax\lib\site-packages\tensorflow\python\data\ops\iterator_ops.py", line 772, in _next_internal
    ret = gen_dataset_ops.iterator_get_next(
  File "F:\python_envirments\waymax\lib\site-packages\tensorflow\python\ops\gen_dataset_ops.py", line 3055, in iterator_get_next
    _ops.raise_from_not_ok_status(e, name)
  File "F:\python_envirments\waymax\lib\site-packages\tensorflow\python\framework\ops.py", line 5888, in raise_from_not_ok_status
    raise core._status_to_exception(e) from None  # pylint: disable=protected-access
tensorflow.python.framework.errors_impl.PermissionDeniedError: {{function_node __wrapped__IteratorGetNext_output_types_31_device_/job:localhost/replica:0/task:0/device:CPU:0}} Error executing an HTTP request: HTTP response code 403 with body '<?xml version='1.0' encoding='UTF-8'?><Error><Code>AccessDenied</Code><Message>Access denied.</Message><Details>Anonymous caller does not have storage.objects.get access to the Google Cloud Storage object. Permission 'storage.objects.get' denied on resource (or it may not exist).</Details></Error>'
	 when reading gs://waymo_open_dataset_motion_v_1_1_0/uncompressed/tf_example/training/training_tfexample.tfrecord-00000-of-01000 [Op:IteratorGetNext] name: 

Process finished with exit code 1

plz help me

justinjfu, 2023-10-31:
Hi @DA21S321D,

Could you try the following diagnostic steps? This would help us write an FAQ for dataset access since we're getting quite a few of these issues raised on github:

(1) Double-check that you can access WOMD via the command line
```
$ gsutil ls gs://waymo_open_dataset_motion_v_1_2_0/uncompressed
gs://waymo_open_dataset_motion_v_1_2_0/uncompressed/lidar/
gs://waymo_open_dataset_motion_v_1_2_0/uncompressed/scenario/
gs://waymo_open_dataset_motion_v_1_2_0/uncompressed/tf_example/
```

(2) Double-check that the default credentials files exist. On Linux, there should be a file called `~/.config/gcloud/application_default_credentials.json`

If that file doesn't exist, you need to run `gcloud auth application-default login`

(3) If you are running in colab, you may need to set `os.environ['GOOGLE_APPLICATION_CREDENTIALS'] =r'.../path/to/application_default_credentials.json'` (see https://github.com/waymo-research/waymax/issues/10)

Let me know if any of these work or help identify the issue.

DA21S321D, 2023-11-01:
thanks, your demo should add some autentication like this( i run properly at local)
(3) If you are running in colab, or if the above doesn't work, you may need to manually set os.environ['GOOGLE_APPLICATION_CREDENTIALS'] =r'.../path/to/application_default_credentials.json' (see https://github.com/waymo-research/waymax/issues/10)

DA21S321D, 2023-11-01:
Hi, i am so confused that even i followed your steps, i still can't run properly. i run on colab here is my error:
[image]
i successfully login and:
[image]

justinjfu, 2023-11-01:
Could you try the instructions on this website? https://www.tensorflow.org/datasets/gcs#simplified_instructions

In particular, add this to your colab:
```
from google.colab import auth
auth.authenticate_user()
```

DA21S321D, 2023-11-02:
[image]
success! Thanks myman!

cjy875428, 2023-11-02:
> Could you try the instructions on this website? https://www.tensorflow.org/datasets/gcs#simplified_instructions
> 
> In particular, add this to your colab:
> 
> ```
> from google.colab import auth
> auth.authenticate_user()
> ```Yes,it works on colab.Tx!#28

cjy875428, 2023-11-02:
> Could you try the instructions on this website? https://www.tensorflow.org/datasets/gcs#simplified_instructions
> 
> In particular, add this to your colab:
> 
> ```
> from google.colab import auth
> auth.authenticate_user()
> ```

Yes!it works on colab.TX for solving the same problem in #28

Jin0932, 2023-12-06:
Hi @justinjfu, I have download the data, but i also met some question when I load dataset, and I hope you can give me some advice,thanks!
[image]

Traceback (most recent call last):
  File "f:\CodewithPaper\waymo\main.py", line 17, in <module>
    scenario = next(data_iter)
  File "C:\Users\Wang\AppData\Local\anaconda3\envs\waymo\lib\site-packages\waymax\dataloader\dataloader_utils.py", line 229, in get_data_generator
    for example in dataset.as_numpy_iterator():
  File "C:\Users\Wang\AppData\Local\anaconda3\envs\waymo\lib\site-packages\tensorflow\python\data\ops\dataset_ops.py", line 4733, in __next__
    return nest.map_structure(to_numpy, next(self._iterator))
  File "C:\Users\Wang\AppData\Local\anaconda3\envs\waymo\lib\site-packages\tensorflow\python\data\ops\iterator_ops.py", line 810, in __next__
    return self._next_internal()
  File "C:\Users\Wang\AppData\Local\anaconda3\envs\waymo\lib\site-packages\tensorflow\python\data\ops\iterator_ops.py", line 773, in _next_internal
    ret = gen_dataset_ops.iterator_get_next(
  File "C:\Users\Wang\AppData\Local\anaconda3\envs\waymo\lib\site-packages\tensorflow\python\ops\gen_dataset_ops.py", line 3024, in iterator_get_next
    _result = pywrap_tfe.TFE_Py_FastPathExecute(
**UnicodeDecodeError: 'utf-8' codec can't decode byte 0xd5 in position 165: invalid continuation byte**

samuela, 2025-03-25:
FWIW I ran into some version of this and following the steps [here](https://github.com/waymo-research/waymax?tab=readme-ov-file#configure-access-to-waymo-open-motion-dataset) worked.

samuela, 2025-05-30:
In the interest of cleaning up stale issues I will close this for now. Feel free to open a new issue if you are still experiencing this!

closed 2025-05-30 by samuela