waymo-research/waymax #33: 3D view of simulator state

cisimon7, 2023-11-02:
Is there a way to change the view of the simulator state and have something like in the waymax paper with the dark theme and colors:
[image]

At the moment, what I am getting seem to be the default settings like below:
[image]

I noticed I can pass a viz_config when calling visualization.plot_simulator_state(...) function, but the config seem not to have any key word for this.

Jin0932, 2023-12-05:
> Is there a way to change the view of the simulator state and have something like in the waymax paper with the dark theme and colors: ![waymax](https://private-user-images.githubusercontent.com/52794824/280035381-60946108-2a82-4411-9e02-717560cb301a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE3NzcwMDcsIm5iZiI6MTcwMTc3NjcwNywicGF0aCI6Ii81Mjc5NDgyNC8yODAwMzUzODEtNjA5NDYxMDgtMmE4Mi00NDExLTllMDItNzE3NTYwY2IzMDFhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA1VDExNDUwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZjYmUzNzg2MWFiY2Q3NTYxZDExMDIzN2FhMjg3YTkzZDE3ODYyMzdjZWExMzRiZGVjZDZiYThlNGQwYTQ0ODUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.5gu38xkBiJQu0faDEIpscpvEEmUGdhEkKT22tqFw918)
> 
> At the moment, what I am getting seem to be the default settings like below: <img alt="" width="300" height="300" src="https://private-user-images.githubusercontent.com/52794824/280035905-ad914c6e-4b2b-4a12-863b-18d26562b8b9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE3NzcwMDcsIm5iZiI6MTcwMTc3NjcwNywicGF0aCI6Ii81Mjc5NDgyNC8yODAwMzU5MDUtYWQ5MTRjNmUtNGIyYi00YTEyLTg2M2ItMThkMjY1NjJiOGI5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA1VDExNDUwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThiNTNmMGNkODI5ODI2YTI5OGVmNjA0NzlhOTcyMjEyYTQ3ZTQyMTY5ZDZmNDcwNThhZDE1NWY5MjQzNzEwODEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.myRHFIp43jl6rSsXwdF7GZWwwdLwkmEv39uD2pPjEB0">
> 
> I noticed I can pass a viz_config when calling visualization.plot_simulator_state(...) function, but the config seem not to have any key word for this.

Hi have you solve this question?

samuela, 2025-06-17:
Hi @cisimon7 , we don't currently support 3d visualizations in the open source release. This should be doable however with Three.js or similar. If you build something like this do let us know! closing for now

closed 2025-06-17 by samuela