waymo-research/waymax #36: A serious bug in PlanningAgentEnvironment?

gaoyinfeng, 2023-11-17:
In line 287 of  `waymax/env/planning_agent_environment.py`, the definition of `action` is:

>       action: The action to apply, of shape (..., num_objects). The
>         actions.valid field is used to denote which objects are being controlled
>         - objects whose valid is False will fallback to default behavior
>         specified by self.dynamics.
But in line 295, the shape of `planning_agent_action ` is  `(..., num_objects, num_objects)` after the update:

>     planning_agent_action = self._planning_agent_dynamics.compute_update(
>         action, state.current_sim_trajectory
>     ).as_action()
Which will cause an AssertionError later in line 328 with `self._state_dynamics.forward`:

>   File "/home/gdg/Waymo/waymax/dynamics/abstract_dynamics.py", line 69, in forward
>     chex.assert_equal(trajectory.shape[:-1], action.shape)
>   File "/home/gdg/anaconda3/envs/waymax_pure/lib/python3.11/site-packages/chex/_src/asserts_internal.py", line 250, in _chex_assert_fn
>     host_assertion_fn(
>   ^^^^^^^^^^^^^^^^^^^^
>   File "/home/gdg/anaconda3/envs/waymax_pure/lib/python3.11/site-packages/chex/_src/asserts_internal.py", line 196, in _assert_on_host
>     raise exception_type(error_msg)
> AssertionError: [Chex] Assertion assert_equal failed: Tuples differ: (32,) != (32, 32)

I define the `planning_agent`as an `expert_actor` which outputs actions to follow log trajectories. I built the program as in the Demos but replaced the `MultiAgentEnvironment(or BaseEnvironment )` with `PlanningAgentEnvironment`, and also defined the controlled object as SDC.
I managed to overcome this issue by directly defining `planning_agent_action=action`, my question is whether this is a bug or is because something goes wrong in my implementation?

justinjfu, 2023-11-27:
Hi,

Thanks for catching this - this is probably a mistake in the documentation. The action shape should match `env.action_spec()`, which lacks the object dimension for PlanningAgentEnvironment. Can you try slicing out the SDC action from your action and trying the same code again? This can be done using `datatypes.select_by_onehot(action, state.object_metadata.is_sdc)`

gaoyinfeng, 2023-12-04:
Hi,

@justinjfu , Thanks for your patient advice, this is much more reasonable and it solves my problem. BTW, I have another question about the training time for RL and IL methods reported in the paper, I believe it is a very time-consuming task to train an agent considering there are more than 500k+ sdc.

closed 2024-09-04 by nicomon24