waymo-research/waymax #40: How to generate a self-designed road network structure?

cjybupt, 2023-11-30:
Hi, I am interested in your simulator for road graph.
After reading the Waymax documentation in https://waymo-research.github.io/waymax/docs/autoapi/waymax/datatypes/roadgraph/index.html, I have known that your simulator supports loading data from Waymo Open Motion Dataset ,loading data from local disk and streaming data remotely via Google Cloud.
However, I am wondering how to build my own map like the example format in https://waymo.com/open/data/motion/tfexample.
Are there any simple methods to generate a map in this format? Is there a third-party software could help me build efficiently?
plz help me,Thx!

Into-sight, 2024-07-07:
Hi man.Have you discovered a solution? I am also actively pursuing that

Mayuemei, 2024-08-03:
I also want to ask how to create a custom road network

Into-sight, 2024-08-08:
hi, you can parse the womd data format and then put your own data into womd_dict format.
I enclose a copy of through (https://github.com/waymo-research/waymax/blob/main/waymax/dataloader/dataloader_utils.py#L63) parsing womd. Please note that this is the format saved by df and not womd at runtime.

PS: 1, this is the representation in df, and the suffix column indicates that its shape in womd is [rows, num].
2. state is easy to complete, but roadgraph is more complicated