waymo-research/waymax #43: [QUESTION] What does state/all/ mean in the Waymo Open Motion Dataset?

JunhongXu, 2023-12-15:
Hello,

Thanks for your effort in making this simulator public. When looking at the `trajectory_from_womd_dict` function in the `womd_factories.py`, I found that the default `time_key` is `all`. However, in https://waymo.com/open/data/motion/tfexample, I only found `past`, `current`, and `future`. My question is, what does the `all` key stand for?

justinjfu, 2023-12-19:
`all` is the concatenation of `past`, `current`, and `future`. 

See: https://github.com/waymo-research/waymax/blob/main/waymax/dataloader/womd_utils.py#L58

JunhongXu, 2023-12-19:
Thanks for your response. If I understand correctly, does Waymax only load the initial conditions of the agents in a trip segment, ignoring agents that will later appear during this trip?

justinjfu, 2023-12-19:
This can be controlled by the `allow_new_objects_after_warmup` flag: https://github.com/waymo-research/waymax/blob/main/waymax/config.py#L207

If you set this to False, it would be the setting you described.

JunhongXu, 2023-12-19:
> This can be controlled by the `allow_new_objects_after_warmup` flag: https://github.com/waymo-research/waymax/blob/main/waymax/config.py#L207
> 
> If you set this to False, it would be the setting you described.

Great! Thanks for your reply!

closed 2023-12-19 by JunhongXu