waymo-research/waymax #51: Waymax visualization

hansungkim98122, 2024-02-07:
Hi, I was wondering how we can visualize the simulation environment in the style shown in this reference picture:

[image]

It seems like the Waymax package only offers more rudimentary visualization of the simulation environment shown below:
[image]

Any help or guidance is appreciated! Thanks.

Jin0932, 2024-05-10:
@hansungkim98122 Hi, have you solve it?

RomanChiva, 2024-08-02:
I have the same issue, any updates?

hansungkim98122, 2024-08-02:
Unfortunately, no. Not sure if it is possible

closed 2024-08-02 by hansungkim98122