waymo-research/waymax #68: Code in tutorial throws AttributeError: 'FigureCanvasTkAgg' object has no attribute 'tostring_rgb'. Did you mean: 'tostring_argb'?

the-big-bad-wolf, 2025-01-17:
I tried running the code in the [Scenario Data Loading tutorial](https://waymo-research.github.io/waymax/docs/notebooks/data_demo.html), but the program crashed.

Traceback (most recent call last):
  File "c:\Users\----\Documents\WaymaxAutoPlanner\main.py", line 16, in <module>
    img = visualization.plot_simulator_state(scenario, use_log_traj=True)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\----\Documents\WaymaxAutoPlanner\.venv\Lib\site-packages\waymax\visualization\viz.py", line 311, in plot_simulator_state
    return utils.img_from_fig(fig)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\----\Documents\WaymaxAutoPlanner\.venv\Lib\site-packages\waymax\visualization\utils.py", line 89, in img_from_fig
    data = np.frombuffer(fig.canvas.tostring_rgb(), dtype=np.uint8)
                         ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'FigureCanvasTkAgg' object has no attribute 'tostring_rgb'. Did you mean: 'tostring_argb'?

vcharraut, 2025-01-17:
Downgrading matplotlib to a version below 3.10 fix the issue

```
pip install matplotlib<3.10
```

the-big-bad-wolf, 2025-01-17:
I uninstalled matplotlib 3.10 and installed matplotlib 3.9.4. 

Now I get
<IPython.core.display.HTML object>
Traceback (most recent call last):
  File "c:\Users\--\Documents\WaymaxAutoPlanner\main.py", line 30, in <module>
    mediapy.show_video(imgs, fps=10)
  File "C:\Users\--\Documents\WaymaxAutoPlanner\.venv\Lib\site-packages\mediapy\__init__.py", line 1862, in show_video
    return show_videos([images], [title], **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\--\Documents\WaymaxAutoPlanner\.venv\Lib\site-packages\mediapy\__init__.py", line 1944, in show_videos
    data = compress_video(
           ^^^^^^^^^^^^^^^
  File "C:\Users\--\Documents\WaymaxAutoPlanner\.venv\Lib\site-packages\mediapy\__init__.py", line 1781, in compress_video
    write_video(tmp_path, images, codec=codec, **kwargs)
  File "C:\Users\--\Documents\WaymaxAutoPlanner\.venv\Lib\site-packages\mediapy\__init__.py", line 1751, in write_video
    with VideoWriter(path, shape=shape, dtype=dtype, **kwargs) as writer:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\--\Documents\WaymaxAutoPlanner\.venv\Lib\site-packages\mediapy\__init__.py", line 1571, in __enter__
    ffmpeg_path = _get_ffmpeg_path()
                  ^^^^^^^^^^^^^^^^^^
  File "C:\Users\--\Documents\WaymaxAutoPlanner\.venv\Lib\site-packages\mediapy\__init__.py", line 1170, in _get_ffmpeg_path
    raise RuntimeError(
RuntimeError: Program 'ffmpeg' is not found; perhaps install ffmpeg using 'apt install ffmpeg'.

I am running Python 3.12 on a Windows machine. Is only Linux supported?

Zhzdpaaa, 2025-01-22:
ValueError: cannot reshape array of size 1440000 into shape (600,600,3)

出错文件：waymax.utils.img_from_fig()

这是因为，waymax的utils.img_from_fig(fig)函数中，在处理图像时，先转成了RGB的一维数组形式，然后再重塑成图像的三维数组（h,w,r,g,b）格式，但我图像转换的函数库版本高，已经弃用了RGB的方法（所以我就换成了ARGB的格式，也就导致了上面的不匹配的错误），使用的是ARGB的格式（就是加了一个透明度），就无法适配指定的（h,w,r,g,b）格式了，所以我就把（h,w,r,g,b）格式改成了（h,w,r,g,b,a）格式，就画出来了，图片上应该是加上了透明度，修改后的代码如下：
def img_from_fig(fig: matplotlib.figure.Figure) -> np.ndarray:
  """Returns a [H, W, 3] uint8 np image from fig.canvas.tostring_rgb()."""
  # Just enough margin in the figure to display xticks and yticks.
  fig.subplots_adjust(
      left=0.08, bottom=0.08, right=0.98, top=0.98, wspace=0.0, hspace=0.0
  )
  fig.canvas.draw()
  data = np.frombuffer(fig.canvas.tostring_argb(), dtype=np.uint8)
  r = data[1::4]  # 红色通道
  g = data[2::4]  # 绿色通道
  b = data[3::4]  # 蓝色通道
  img = np.stack([r, g, b], axis=-1).reshape(fig.canvas.get_width_height()[::-1] + (3,))
  plt.close(fig)
  return img

Zhzdpaaa, 2025-01-22:
ValueError: cannot reshape array of size 1440000 into shape (600,600,3)

出错文件：waymax.utils.img_from_fig()

这是因为，waymax的utils.img_from_fig(fig)函数中，在处理图像时，先转成了RGB的一维数组形式，然后再重塑成图像的三维数组（h,w,r,g,b）格式，但我图像转换的函数库版本高，已经弃用了RGB的方法（所以我就换成了ARGB的格式，也就导致了上面的不匹配的错误），使用的是ARGB的格式（就是加了一个透明度），就无法适配指定的（h,w,r,g,b）格式了，所以我就把（h,w,r,g,b）格式改成了（h,w,r,g,b,a）格式，就画出来了，图片上应该是加上了透明度，修改后的代码如下：

`def img_from_fig(fig: matplotlib.figure.Figure) -> np.ndarray:
  """Returns a [H, W, 3] uint8 np image from fig.canvas.tostring_rgb()."""
  # Just enough margin in the figure to display xticks and yticks.
  fig.subplots_adjust(
      left=0.08, bottom=0.08, right=0.98, top=0.98, wspace=0.0, hspace=0.0
  )
  fig.canvas.draw()
  data = np.frombuffer(fig.canvas.tostring_argb(), dtype=np.uint8)
  r = data[1::4]  # 红色通道
  g = data[2::4]  # 绿色通道
  b = data[3::4]  # 蓝色通道
  img = np.stack([r, g, b], axis=-1).reshape(fig.canvas.get_width_height()[::-1] + (3,))
  plt.close(fig)
  return img`

the-big-bad-wolf, 2025-01-22:
You are using the latest version of the dataset right? I got the same error when trying to use version 1.3.0, but downgrading to the same version used in the tutorial fixed it.

Zhzdpaaa, 2025-01-22:
是的，解决这个问题，有两种方法：降低版本或者使用现在这个版本的ARGB的格式（即修改waymax.utils.img_from_fig()中的fig.canvas.tostring_rgb()为fig.canvas.tostring_argb()），但是直接修改会带来新的问题，即我上面给出的问题ValueError: cannot reshape array of size 1440000 into shape (600,600,3)，数据大小不匹配。
这是因为修改为ARGB的格式之后，图片的大小是（600，600，4）（乘一下正好是1440000），然而waymax.utils.img_from_fig()的后续代码中img = data.reshape(fig.canvas.get_width_height()[::-1] + (3,))，这里给出的目标格式就是(600,600,3)
所以，解决这个问题，也有两个方法：
1、修改为img = data.reshape(fig.canvas.get_width_height()[::-1] + (4,))，即把目标格式改为(600,600,4)，但这样会在生成的图像上加入透明度的信息，看起来粉粉的
2、使用ARGB的格式提取后，去除透明度的信息，这样就可以达到原本的效果，如下修改即可
data = np.frombuffer(fig.canvas.tostring_argb(), dtype=np.uint8)
  r = data[1::4]  # 红色通道
  g = data[2::4]  # 绿色通道
  b = data[3::4]  # 蓝色通道
  img = np.stack([r, g, b], axis=-1).reshape(fig.canvas.get_width_height()[::-1] + (3,))

Zhzdpaaa, 2025-01-22:
是的，我使用的是新的版本，解决这个问题，有两种方法：降低版本或者使用现在这个版本的ARGB的格式（即修改waymax.utils.img_from_fig()中的fig.canvas.tostring_rgb()为fig.canvas.tostring_argb()），但是直接修改会带来新的问题，即我上面给出的问题ValueError: cannot reshape array of size 1440000 into shape (600,600,3)，数据大小不匹配。
这是因为修改为ARGB的格式之后，图片的大小是（600，600，4）（乘一下正好是1440000），然而waymax.utils.img_from_fig()的后续代码中img = data.reshape(fig.canvas.get_width_height()[::-1] + (3,))，这里给出的目标格式就是(600,600,3)
所以，解决这个问题，也有两个方法：
1、修改为img = data.reshape(fig.canvas.get_width_height()[::-1] + (4,))，即把目标格式改为(600,600,4)，但这样会在生成的图像上加入透明度的信息，看起来粉粉的
2、使用ARGB的格式提取后，去除透明度的信息，这样就可以达到原本的效果，如下修改即可
data = np.frombuffer(fig.canvas.tostring_argb(), dtype=np.uint8)
r = data[1::4] # 红色通道
g = data[2::4] # 绿色通道
b = data[3::4] # 蓝色通道
img = np.stack([r, g, b], axis=-1).reshape(fig.canvas.get_width_height()[::-1] + (3,))

Zhzdpaaa, 2025-01-22:
Yes, I am using the new version. To solve this issue, there are two methods: one is to downgrade the version, and the other is to use the ARGB format in the current version (i.e., modify `waymax.utils.img_from_fig()` by changing `fig.canvas.tostring_rgb()` to `fig.canvas.tostring_argb()`). However, directly modifying it leads to a new problem, specifically the error `ValueError: cannot reshape array of size 1440000 into shape (600,600,3)`, due to data size mismatch.

This is because, after changing to the ARGB format, the image size becomes (600,600,4) (which gives 1440000 when multiplied), but in the subsequent code of `waymax.utils.img_from_fig()`, `img = data.reshape(fig.canvas.get_width_height()[::-1] + (3,))` is used, where the target format is specified as (600,600,3).

Therefore, to solve this issue, there are two approaches:
1. Modify it to `img = data.reshape(fig.canvas.get_width_height()[::-1] + (4,))`, changing the target format to (600,600,4), but this will add transparency information to the generated image, making it appear pinkish.
2. After extracting the ARGB format, remove the transparency information to restore the original effect. You can modify it as follows:
```python
data = np.frombuffer(fig.canvas.tostring_argb(), dtype=np.uint8)
r = data[1::4]  # Red channel
g = data[2::4]  # Green channel
b = data[3::4]  # Blue channel
img = np.stack([r, g, b], axis=-1).reshape(fig.canvas.get_width_height()[::-1] + (3,))
```

Zhzdpaaa, 2025-01-23:
Correct it, the error file is: waymax-main/waymax/visualization/utils.py

> ValueError: cannot reshape array of size 1440000 into shape (600,600,3)
>
> 出错文件：waymax.utils.img_from_fig() [...]

Zhzdpaaa, 2025-01-23:
Correct it, the error file is: waymax-main/waymax/visualization/utils.py中的img_from_fig(fig)函数

> ValueError: cannot reshape array of size 1440000 into shape (600,600,3)
>
> 出错文件：waymax.utils.img_from_fig() [...]

closed 2025-01-25 by the-big-bad-wolf