waymo-research/waymax #88: Cannot load the data using dataloader following the quickstart guide.

siddhi47, 2025-07-14:
I am trying to load dataset into my local machine using the following code.
```
import numpy as np
import mediapy
from tqdm import tqdm
import dataclasses

from waymax import config as _config
from waymax import dataloader
from waymax import datatypes
from waymax import visualization

config = dataclasses.replace(_config.WOD_1_1_0_TRAINING, max_num_objects=32)
data_iter = dataloader.simulator_state_generator(config=config)
scenario = next(data_iter)
```

- I have made sure that I logged in using the same google account.
- Google cloud is properly authenticated and logged in.

This is the error that i get:

```
Traceback (most recent call last):
  File "/Users/siddhibajracharya/personal/git/waymax/a.py", line 14, in <module>
    scenario = next(data_iter)
               ^^^^^^^^^^^^^^^
  File "/Users/siddhibajracharya/personal/git/waymax/waymax/dataloader/dataloader_utils.py", line 229, in get_data_generator
    for example in dataset.as_numpy_iterator():
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/siddhibajracharya/personal/git/waymax/waymo/lib/python3.12/site-packages/tensorflow/python/data/ops/dataset_ops.py", line 4788, in __next__
    return nest.map_structure(to_numpy, next(self._iterator))
                                        ^^^^^^^^^^^^^^^^^^^^
  File "/Users/siddhibajracharya/personal/git/waymax/waymo/lib/python3.12/site-packages/tensorflow/python/data/ops/iterator_ops.py", line 826, in __next__
    return self._next_internal()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/siddhibajracharya/personal/git/waymax/waymo/lib/python3.12/site-packages/tensorflow/python/data/ops/iterator_ops.py", line 776, in _next_internal
    ret = gen_dataset_ops.iterator_get_next(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/siddhibajracharya/personal/git/waymax/waymo/lib/python3.12/site-packages/tensorflow/python/ops/gen_dataset_ops.py", line 3086, in iterator_get_next
    _ops.raise_from_not_ok_status(e, name)
  File "/Users/siddhibajracharya/personal/git/waymax/waymo/lib/python3.12/site-packages/tensorflow/python/framework/ops.py", line 6006, in raise_from_not_ok_status
    raise core._status_to_exception(e) from None  # pylint: disable=protected-access
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
tensorflow.python.framework.errors_impl.InvalidArgumentError: {{function_node __wrapped__IteratorGetNext_output_types_31_device_/job:localhost/replica:0/task:0/device:CPU:0}} GCS path doesn't contain a bucket name: gs:///waymo_open_dataset_motion_v_1_1_0/uncompressed/tf_e
xample/training/training_tfexample.tfrecord-00000-of-01000 [Op:IteratorGetNext] name: 
```

These are my configuration;
Laptop Macbook M2 Air
Python 3.12.3
MacOS Version 15.5 (24F74)

Pip packages:
```
Package                 Version
----------------------- -----------
absl-py                 2.3.1
asttokens               3.0.0
astunparse              1.6.3
attrs                   25.3.0
certifi                 2025.7.14
charset-normalizer      3.4.2
chex                    0.1.89
contourpy               1.3.2
cycler                  0.12.1
decorator               5.2.1
dm-env                  1.6
dm-tree                 0.1.9
etils                   1.12.2
executing               2.2.0
flatbuffers             25.2.10
flax                    0.10.7
fonttools               4.58.5
fsspec                  2025.5.1
gast                    0.6.0
google-pasta            0.2.0
grpcio                  1.73.1
h5py                    3.14.0
humanize                4.12.3
idna                    3.10
immutabledict           4.2.1
importlib-resources     6.5.2
ipython                 9.4.0
ipython-pygments-lexers 1.1.1
jax                     0.6.2
jaxlib                  0.6.2
jedi                    0.19.2
keras                   3.10.0
kiwisolver              1.4.8
libclang                18.1.1
markdown                3.8.2
markdown-it-py          3.0.0
markupsafe              3.0.2
matplotlib              3.10.3
matplotlib-inline       0.1.7
mdurl                   0.1.2
mediapy                 1.2.4
ml-dtypes               0.5.1
msgpack                 1.1.1
namex                   0.1.0
nest-asyncio            1.6.0
numpy                   2.1.3
opt-einsum              3.4.0
optax                   0.2.5
optree                  0.16.0
orbax-checkpoint        0.11.19
packaging               25.0
parso                   0.8.4
pexpect                 4.9.0
pillow                  11.3.0
prompt-toolkit          3.0.51
protobuf                5.29.5
ptyprocess              0.7.0
pure-eval               0.2.3
pygments                2.19.2
pyparsing               3.2.3
python-dateutil         2.9.0.post0
pyyaml                  6.0.2
requests                2.32.4
rich                    14.0.0
scipy                   1.16.0
setuptools              80.9.0
simplejson              3.20.1
six                     1.17.0
stack-data              0.6.3
tensorboard             2.19.0
tensorboard-data-server 0.7.2
tensorflow              2.19.0
tensorstore             0.1.76
termcolor               3.1.0
toolz                   1.0.0
tqdm                    4.67.1
traitlets               5.14.3
treescope               0.1.9
typing-extensions       4.14.1
urllib3                 2.5.0
waymo-waymax            0.1.0
wcwidth                 0.2.13
werkzeug                3.1.3
wheel                   0.45.1
wrapt                   1.17.2
zipp                    3.23.0

```

alexyin1, 2025-07-15:
I ran into the same issue with the tutorial notebooks; Perhaps the GCP link has changed? The below works for me
```
config = _config.DatasetConfig(
    path="gs://waymo_open_dataset_motion_v_1_3_0/uncompressed/tf_example/training/training_tfexample.tfrecord-00000-of-01000",
    max_num_objects=32
)
```
^ i got the URL from the gcp GUI, and choosing "Copy gsutil URI" https://console.cloud.google.com/storage/browser/waymo_open_dataset_motion_v_1_3_0

siddhi47, 2025-08-04:
@alexyin1 Thanks. It worked for me too. 
Adding this line in the config worked for me:

```python
#waymax.config.py
WOD_1_3_0_TEST = DatasetConfig(
    path="gs://waymo_open_dataset_motion_v_1_3_0/uncompressed/tf_example/training/training_tfexample.tfrecord@1000",
    data_format=DataFormat.TFRECORD,
)
```

In main script:

```python
config = dataclasses.replace(_config.WOD_1_3_0_TRAINING, max_num_objects=32)
data_iter = dataloader.simulator_state_generator(config=config)
scenario = next(data_iter)
```